miguel5612/ThermistorLibrary #8: Error compiling for ESP8266

vangalvin, 2020-07-19:
I tried to compile this for an ESP8266 and get the following errors.

WARNING: library ThermistorLibrary-master claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s).
C:\Arduino\libraries\ThermistorLibrary-master\src\thermistor.cpp: In member function 'float thermistor::analog2temp()':

C:\Arduino\libraries\ThermistorLibrary-master\src\thermistor.cpp:41:38: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]

     for (i = 1; i < heater_ttbl_map[e]; i++) {

Any sugestions as to how to fix this?

Otherwise the library works a treat on the NANO, however i need to report the temperature back via WiFi and the ESP seemed to be the easier option for this.

bjkayani, 2020-11-18:
I am facing the same issue. Did you find any solution?

ondrejnovakcvut, 2021-02-26:
I've fixed this by replacing `heater_ttbl_map` by `heater_ttbllen_map`

gustavya, 2021-06-12:
What about ESP32, any way to make this work? thanks

closed 2022-03-19 by miguel5612

PsyonicOne, 2023-06-15:
I would also like to know how to get this library working on ESP32. I've tried getting thermistor.cpp to read directly from `raw += adc1_get_raw(ADC1_CHANNEL_5)` instead of `raw += analogRead(_pin);` but I always get 0.00C as the result. Printing `raw` seems to give sane values...

PsyonicOne, 2023-06-15:
I figured out how to get the ESP32 working... Need to set `adc1_config_width(ADC_WIDTH_BIT_10)`
It would be nice if this worked with other bit resolutions

PR #11 (2022-03-19): Fix #8
This fixes #8. Thermistor table contents were used instead of thermistor table length to bound the for loop.
commit 09a9dad9 (2022-03-19): Fix #8
commit ba7d0330 (2022-03-19): Merge pull request #11 from ondrejnovakcvut/fix-issue-8

Fix #8