miguel5612/ThermistorLibrary #14: All sensors use  table for TEMP_SENSOR_0

bondematt, 2022-02-04:
As the title says all sensors are using the table defined as TEMP_SENSOR_0

I changed the definition in configuration.h to:
#define TEMP_SENSOR_0 80 // This is thermistor 0
#define TEMP_SENSOR_1 4 // This is thermistor 1

My definitions in my sketch are:
thermistor therm1(A7,0);  //Connect thermistor on A7, 0 represents TEMP_SENSOR_0 ( configuration.h for more)
thermistor therm2(A3,1);  //Connect thermistor on A3, 1 represents TEMP_SENSOR_1 ( configuration.h for more)

and I'm calling:
temp1 = (int)round(therm1.analog2temp()); // read temperature
temp2 = (int)round(therm2.analog2temp()); // read temperature

I've tried 2 and 3 for the second thermistor and changing the table for the corresponding TEMP_SENSOR_X but no luck. When I call thermX.analog2temp() it always returns using the TEMP_SENSOR_0 table. I'm using both a 10K Ohm 3950 and 100K Ohm 3950 so I'm hoping to have two tables, but for now I'm going to just use 100K for both

Setting TEMP_SENSOR_0 to 998 or 999 returns the dummy table 25C/100C for both sensors. Calling it with it set to 80 gives me the correct value on therm1(100K) and a very high value on therm2(10K)

miguel5612, 2022-03-20:
Remember that in addition to modifying configuration.h you must modify the sensor number in your program 
thermistor therm1(A0,0); // Analog Pin which is connected to the 3950 temperature sensor, and 0 represents TEMP_SENSOR_0 (see configuration.h for more information).

Example in this case the 0 means that you are using the TEMP_SENSOR_0 pin configuration.

closed 2022-03-20 by miguel5612

bondematt, 2022-03-20:
Hi Miguel, please see the definitions I am using in my sketch above. This issue can be summed up as: No matter what sensor number you put in the thermistor definition in the sketch, it always used the configuration.h for TEMP_SENSOR_0

bondematt, 2022-03-20:
For more clarity, thermistor therm2(A3,1) always uses the table defined by TEMP_SENSOR_0, doesn't matter if I define it as (A3,1)  to(A3,5) it uses TEMP_SENSOR_0

skotl, 2023-11-09:
I'm agreeing with @bondematt, @miguel5612 - no matter which sensor ID is specified, it always uses the type from `TEMP_SENSOR_0`

My initialiser inside my caller's constructor:

```csharp
 thermistor therm(PIN_TEMPERATURE, 1);
```

and my `configuration.h`:

```csharp
#define TEMP_SENSOR_0 2 // This is thermistor 0
#define TEMP_SENSOR_1 51 // This is thermistor 1
#define TEMP_SENSOR_2 0 // This is thermistor 2
#define TEMP_SENSOR_3 0 // This is thermistor 3
#define TEMP_SENSOR_4 0 // This is thermistor 4
#define TEMP_SENSOR_BED 80 // This is not used thermistor
```

No matter what I set the `sensorNumber` to in the call to `therm()`, it will always use the thermistor definition attached to `TEMP_SENSOR_0`